per1234/Ariadne-Bootloader #13: Use EEPROM flag to skip bootloader

per1234, 2016-02-05:
Ariadne currently runs the bootloader on every reset. The `img_stat` EEPROM address is currently used to disable the bootloader timeout so that it will perpetually wait for an upload before running the program. At the start of each upload the bad flag value is written to the address and then the OK flag value is only written to it after successful upload(and currently I think it also verifies the file although that behavior may be removed soon).

The change I propose is that if the `img_stat`EEPROM address is set to the skip flag value in the sketch then on the next reset the bootloader would write the OK value to the address and then immediately time out without waiting for an upload. Define the 3 flag values(OK, bad, and skip) of `img_stat` in neteeprom.h and these defines will also be available to the user in the copy of that file included in the bundled network configuration library. Add these names to the keywords.txt of that library.

hachi, 2018-01-14:
If I might suggest, having a configurable delay in the eeprom as the number of seconds might be useful. 0 can imply no delay and -1 (255) could imply wait forever.

per1234, 2018-01-14:
Do you think users have significant variation in their delay length needs? I use a script to send the reset command and then start the upload immediately after so I can get by with a very short delay. If someone is doing this manually they will require more time.

I did end up reducing the delay from 20 seconds to 10 seconds in the parent repository (https://github.com/loathingKernel/ariadne-bootloader/commit/e1e3d111fe037cb800b65938698d915b21e2dd7c) but have not gotten around to merging that change (or other significant improvements) to this repository.

My idea here was that if the bootloader had a mechanism to skip the delay then a long bootloader delay won't pose any problems since it will only be encountered when an upload is intended and so a reasonable one-size-fits-all delay would be fine. However, that does require the users to set the EEPROM flag.

There is some relevant discussion here: https://github.com/loathingKernel/ariadne-bootloader/issues/8#issuecomment-222351761
Moving the configuration to flash memory as proposed in that issue would generally be a significant improvement. However, it would make setting this flag less user friendly since it's quite easy and common to write to EEPROM, less so to write to flash from the application.