per1234/Ariadne-Bootloader #17: Allow serial upload if W5x00 is not attached

per1234, 2016-02-09:
If Ethernet controller is not detected then still allow serial upload and use a shorter bootloader delay. This is claimed to have been done already in the [NetLoad bootloader](https://github.com/d0ughb0y/NetLoad).

Update the **Troubleshooting** section of the documentation.

hagaigold, 2017-06-07:
I done it (again) in a bit different way: the decision is base on initialize failure in sockInit() function.
please see bedf5fd0daaad320a1928f1b9c7f86279cbd7fd0

also i think it's should titled as a bug and not as an enhancement. reason is that serial upload is part of that boot-loader, and it fail to work on some conditions.

per1234, 2017-06-07:
Thanks! I'm very excited for this. I think the serial upload only working with the Ethernet controller connected was a very common source of confusion for people. It's even gotten me a couple of times and I should have known better. It's going to take me a little while to get the spare time to test this out but I'll definitely merge it as soon as possible.

hagaigold, 2017-06-07:
you will need this as well 539101d9a127026b9aae7d83cbecca107e054f98 for testing it.
or just look on my develop branch.