per1234/Ariadne-Bootloader #22: Name change

per1234, 2016-02-20:
This repository will continue to change significantly from the original Ariadne(especially after https://github.com/per1234/Ariadne-Bootloader/issues/11) so it is probably best to change the name to avoid confusion. I offered to push my changes back to the main repo in October 2015 but haven't received any response and my vision for this project may differ from theirs anyway. Check if GitHub will redirect the current URL to package_codebendercc_ariadne-bootloader_index.json and if so then I can create a new JSON file for my new name and keep the old one to still allow installation of the original Ariadne via Boards Manager and move all versions except 1.0.0 to the new JSON file.

ChrisFredriksson, 2018-07-11:
Is this fork still active? Seems there are no responses in other "threads". I hope its alive still! :)

per1234, 2018-07-11:
I'm kind of undecided about what will happen with this fork.

Some years ago I started an initiative to add Boards Manager installation support to Ariadne. At that time the author of Ariadne, LoathingKernel, was not going to be able to work on the project for some time and so they requested that I just create a fork.

Since the Ariadne project had already not been active for some time before that I got the feeling it was somewhat abandoned and so I kind of went my own direction with this fork. Then LoathingKernel came back and there has been some really great work done on the original Ariadne project.

I would much prefer there to just be a single focus for this work but I also think that Boards Manager installation support is very important and have not had success moving that forward in the parent repository. I also have a somewhat different vision in that I would like to offer a lot of configuration options and support for as many parts as possible, even if that makes things more complicated and less thoroughly tested. LoathingKernel's philosopy is to provide less options so that it is less complicated and possible to test more thoroughly.

So now I have a situation where there is some good work in this repo that has not been merged back to the parent and some very important work in the parent that has not been merged here.

I'm currently using this version of Ariadne in my own devices but I'm planning to make a transition from the W5100 to the W5500 soon. That chip is currently only supported by the parent so I'm going to need to make the decision of whether I want to continue development of this fork.

ChrisFredriksson, 2018-07-11:
Thanks for the quick response! Well, I've been trying off an on for the last year to get any Ariadne to work on my Controllino MAXI (Mega2560/W5100) and with some great ideas from jkroeze over on the LoathingKernel project and lots of time testing and trying, I just got it to work yesterday - somewhat..

When using the LoathingKernel project I get network access and I can ping the device, but it does not respond to tftp connections.. It does not work with serial upload either.

With your Ariadne project I do get a response and I can upload binaries via tftp, however those programs are never started after the upload is completed.. Also, serial upload is not working.

For both of the above scenarios I've used a simple blink sketch that is ~3kbyte in size.

So I'm not really sure where to go next.. jkroeze is probably on vacation on the LoathingKernel issueboard and no one else seem to have any issues with these bootloaders it seems, at least not the same kind of issues ;)

When it comes to functionality and progression of the code, I'm just happy if it works right now. But some of the features/ideas that you have presented such as the skip bootloader with eeprom flag, I think  its really great and I believe that would be a nice option to have.

Sure, its easier to test and verify that the bootloader is working as good as it possible can if its fewer parts to debug.. but the bootloader could evolve over time and each new "module" you add you can bugtest as much as possible, no need to have it all completed at once - and I'm sure that's not what you meant either.

From what I can tell it doesn't seem to be much that is needed to add W5200 and W5500 support as its already done in the LoathingKernel version.

But of course, if everyone joins one single project that could probably benefit everyone. Thanks to Github I guess that its quite easy to fork off every now and then and add those extra features you're thinking about.. just some small patchwork to the existing code? :)

per1234, 2018-07-11:
For support it's probably best to start a thread on the Arduino Forum. If your problem ends up being caused by a bug then that should be reported to the issue tracker.

I can tell you that I use this repository's latest version on an ATmega2560 with W5100 and have no problems at all. I did some testing of the official Ariadne after the ATmega1284P and W5200/W5500 support was added and didn't have any problems but I haven't tested after my last commit in that repository and I believe LoathingKernel did a couple of commits after that.

Regarding bug testing: The problem is the multiplier effect of each additional configuration option. You end up with math like 10 microcontrollers * 3 network controllers * 5 pinouts * 3 clock options * 3 CS pins * 2 blink options * 2 EEPROM options. That's not the exact equation but you get the idea. Completely testing every possible combination of even the existing settings is extremely time consuming. That can discourage to development work and making production releases. I can easily set up automated full coverage compilation testing but just the fact that the code compiles is no guarantee it actually works. Complete hardware-in-loop testing would be difficult and expensive to automate. So realistically it comes down to the decision of whether it's better to leave some of the testing of the fringe cases to the users or just not support those fringe cases at all. I lean toward the former but that can mean a bad experience for someone just trying to get their project going.

ChrisFredriksson, 2018-07-12:
I've never had any luck with the Arduino forum, you're always pointed out as an idiot for not providing the answer before you even ask a question. So I just can't bother with that forum :(

Yeah, I'm going to try and see if I can fix this problem I'm having somehow.. With LoathingKernel I get no TFTP connection and with your version I do get the file uploaded, but not started.

```
tftp> Verbose mode on.
tftp> Packet tracing on.
tftp> Connected to 192.168.1.128.
Mode: octet Verbose: on Tracing: on
Rexmt-interval: 5 seconds, Max-timeout: 25 seconds
tftp> putting upload.bin to 192.168.1.128:upload.bin [octet]
sent WRQ <file=upload.bin, mode=octet>
received ACK <block=0>
sent DATA <block=1, 512 bytes>
received ACK <block=1>
sent DATA <block=2, 512 bytes>
received ACK <block=2>
sent DATA <block=3, 512 bytes>
received ACK <block=3>
sent DATA <block=4, 512 bytes>
received ACK <block=4>
sent DATA <block=5, 388 bytes>
received ACK <block=5>
Sent 2436 bytes in 4.0 seconds [4872 bits/sec]
```

But it just doesn't start the application afterwards. I can't even enable debugging in any of the bootloaders, it just spits out garbage, no matter what baud I set the terminal to, even at 115200.

I understand, didn't think of that complexity for bug testing ;)